sifive/freedom #132: Scala File in freedom/src/main/scala/unleashed/

wwMark, 2019-09-01:
Could someone give me a short explanation about the four .scala files in  freedom/src/main/scala/unleashed/ and  freedom/src/main/scala/unleashed/vera? I don't have any idea of how are the components of the FU500 connected with each other. Thank you.

erikdanie, 2019-09-03:
Sure. First thing to know is that the freedom repo contains 3 main "classes" of builds, one being the FE300, which is not part of what you are looking at here.
The other two are the FU500 and the IOFPGA. 

IOFPGA is a build designed to go onto an FPGA in order to utilize that FPGA exclusively for IO for another system. For example, a HiFive Unleashed board can be connected to the FMC connector of the FPGA and use the PCIe slot on the FPGA for a graphics card. These transactions are done via Chiplink. That is the src/main/scala/unleashed/IOFPGADesign.scala file. The src/main/scala/unleashed/vera/iofpga/FPGAChip.scala is an older file that was used specifically for the MicroSemi PolarFire FPGA, though in newer builds, it utilizes the main file I mentioned before.

For the FU500, the files of note are /src/main/scala/unleashed/DevKitConfigs.scala and src/main/scala/unleashed/DevKitFPGADesign.scala, which I will refer to as the Config and the Design. The config is used to describe what a build of the FU500 should look like, making use of the API's defined in rocket-chip. You can see that the main thing used here is called BaseConfig, which you can find in rocket-chip, and that the rest simply add on to that. For example, the file does things like set up parameters for the peripherals, adds an error device, adds jtag, etc.
The Design file then takes the information described in the Config and the FPGAShell (imagine it as a wrapper around the top level that contains info and data relevant to the specific FPGA that the design is going on) and connects things at a system level. For example, you can see some of the IOs being driven here (i.e. line 142) and the design querying the shell in order to decide what to connect its busses to (i.e. line 105 checks if the shell has defined a pcie device and if so, it places it and connects it to the system bus)

I hope this makes sense as a cursory-level explanation. Let me know if you have more questions or want to go more in depth about specific lines.

wwMark, 2019-10-11:
@erikdanie Thank you, your reply is of great help!

closed 2019-10-11 by wwMark