sifive/freedom #134: vc707-iofpga and vc707-u500devkit architectural differences?

salmansheikh, 2019-09-07:
What is the difference between the two from an architectural SOC viewpoint? And how can I find the memory map for what peripherals are created for each file?  Also how to

erikdanie, 2019-09-09:
The vc707-u500devkit is a full devkit SoC built around the u500 core. This includes ips for peripherals like GPIO, SPI, UART etc, as well as xilinx ips like PLL, DDR and PCIe controllers. You can find the memory map in the .dts file that gets generated in the builds directory

The vc707-iofpga is a design that makes use of the vc707 as a hub for ios, mainly intended for use with the HiFive Unleashed board. It uses the FMC connector to connect to the board, and then contains a kind of empty shell that connects peripherals like ddr and pcie for the hifive unleashed to make use of, as if they were all peripherals on the same board.

salmansheikh, 2019-09-10:
So, the u500devkit version is intended for booting Linux? I wanted to use bare-metal OS or put an RTOS on it (RTEMS or FreeRTOS) for my test application.

erikdanie, 2019-09-10:
The u500devkit can be used to boot linux, and contains boot code that read in from the sd card on the vc707, expecting to find a linux bbl, loads it directly into ddr and begins executing from there. You can use this to put a bare metal OS, depending on its size, you may want to make use of this sd boot, or the code can be changed to boot straight to a location in memory