sifive/freedom #135: SPI flash programmin file : download link not working

bagyavinoth, 2019-09-13:
Hello , 
i'm trying to setup my arty 7 board for FE310 development. 
I know that we can build the fpga bitstreams for FE310 from source.
But I wanted to get the pre-built image for FE310 from sifive to directly flash & try it in Arty 7 board. Where can i get the pre-built image ? 
Thanks in advance.

Regards,
Vinoth

erikdanie, 2019-09-16:
Hi,
I don't think we have the bitstream readily available online, but you can clone the freedom repo here, follow the readme instructions and run 
> make -f Makefile.e300artydevkit mcs
That should generate the fpga bitstream and mcs file

bagyavinoth, 2019-09-17:
> make -f Makefile.e300artydevkit mcs

Thanks @erikdanie !
I would give it try & get back if i have questions. 

Regards,
Vinoth