sifive/freedom #138: How to insert a delay module between the CPU and memory controller?

wwMark, 2019-10-13:
Hello, could someone give me a clue about how to implement a delay module to delay memory request signal sent from CPU to memory controller? Which source files should I edit and how to build a module that function correctly with the others?

ErsinCK, 2019-10-29:
Interesting. I am trying to do the same thing. Right now, I am modifying this https://github.com/sifive/fpga-shells/blob/14297af2878dc648ffd5751010fa72094ff444b0/src/main/scala/devices/xilinx/xilinxvc707mig/XilinxVC707MIG.scala 

Basically, I am placing registers in between the AXI interface and the memory. But, when I do that. The boot gets stuck. My guess is that I am messing with the synchronization somehow.

wwMark, 2019-10-31:
Hello ErsinCK, i have already solved this issue. You could check this link https://community.arm.com/developer/ip-products/processors/f/cortex-a-forum/6553/can-we-delay-read-and-write-transactions-axi4-by-providing-delay-in-register-slice. It