sifive/freedom #141: Build error when I bump up rocket-chip to master

kenzhang82, 2019-10-20:
Hi there,

I am working on a forked version of Freedom repo, I am trying to get the upstream update from rocket-chip so I bump the rocket-chip to master (hash: f5dd5b09c7eadfb542ccfb740894d78c5a4a1b61).

However I am getting the following error when I am trying to build an e300 platform on FPGA:
```
[error] /home/ken/freedom/rocket-chip/src/main/scala/package.scala:6:16: not found: value chipsalliance
[error]   val config = chipsalliance.rocketchip.config
[error]                ^
[error] /home/ken/freedom/rocket-chip/src/main/scala/amba/ahb/AHBLite.scala:10:29: not found: type Parameters
[error] class AHBLite()(implicit p: Parameters) extends LazyModule {
```
I understand that the current freedom tagged to a very specific version of rocket-chip, I would be appreciated if anyone gives me any insight on this. Thanks!

Ken

erikdanie, 2019-10-22:
Hi Ken,

You need to do a `git submodule update --init` in rocket-chip, since there are new submodules that did not exist in the older version.

kenzhang82, 2019-10-22:
Thanks @erikdanie, I am pretty sure I did this, the following is the output from `git submodule status`

```
-d619ca850846d2ec36da64bf8a28e7d9a3d9ed1b api-config-chipsalliance
 e1aa5f3f5c0cdeb204047c3ca50801d9f7ea25f1 chisel3 (2019-03-11-SNAPSHOT-48-ge1aa5f3)
 228c9a4b7432ac52178d63b8f27fe064aec71e9c firrtl (2019-03-11-SNAPSHOT-25-g228c9a4)
 3ca9cc7901b91ddccf3fcfd686aa7a93ed48d84f hardfloat (v1.2.0-RC1~1)
 77195ab12aefc373ca688e0a9c4d710c13191341 torture (remotes/origin/add_loops-4-g77195ab)
```

kenzhang82, 2019-10-22:
Sorry I was mistaken, that was actually the issue. It is all good now with: `git submodule update --init`

closed 2019-10-22 by kenzhang82

yintianyu, 2019-10-31:
Hey, did you face any other error after bumping Rocket-Chip? As I opened #143