sifive/freedom #142: openocd configuration file for vcu118

shirlynan, 2019-10-29:
I try to debug freedom with openocd on the board vcu118 but could not find configuration file for the board. Has anyone tried that? I have no idea where to get this file ... Thanks a lot.

erikdanie, 2019-10-31:
Hi,

The same one for vc707 or arty will work more or less the same, the main difference being the spi flash.

https://github.com/sifive/freedom-e-sdk/blob/v201908-branch/bsp/freedom-e310-arty/openocd.cfg#L20 

If you don't have or want to use the spi flash, you can just comment out both the lines that start with flash

shirlynan, 2019-11-02:
> Hi,
>
> The same one for vc707 or arty will work more or less the same, the main difference being the spi flash. [...]

Thanks for your reply. Is it also possible to debug riscv without freedom-sdk but using xilinx usb jtag?

erikdanie, 2019-11-06:
Not as of now, but that is something we are looking into

closed 2019-11-09 by shirlynan