sifive/freedom #143: Error after bumping rocket-chip to a little bit newer

yintianyu, 2019-10-31:
In order to fit sifve-inclusivecache, I bumped rocket-chip newer to [bcb3d0a](https://github.com/chipsalliance/rocket-chip/tree/bcb3d0a9c097a4d66687bcf7aded3f94048c6a2e)

However, when I build, it prompted that HasPeripheryMaskROMSlave in DevKitFPGADesign.scala not found.
I find out that MaskROM has helper object attach instead of trait since [#1698](https://github.com/chipsalliance/rocket-chip/pull/1698)

Then I deleted that line, and another error surfaces:
```
[error] (run-main-0) java.lang.IllegalArgumentException: requirement failed: in_xbar.node (A nexus node with parent 'in_xbar/pbus/topMod/topDesign/top' at  (PeripheryBus.scala:36:29)) has 1 required outputs and no possible inputs
[error] java.lang.IllegalArgumentException: requirement failed: in_xbar.node (A nexus node with parent 'in_xbar/pbus/topMod/topDesign/top' at  (PeripheryBus.scala:36:29)) has 1 required outputs and no possible inputs
[error] 	at scala.Predef$.require(Predef.scala:277)
[error] 	at freechips.rocketchip.diplomacy.MixedNexusNode.resolveStar(Nodes.scala:513)
[error] 	at freechips.rocketchip.diplomacy.MixedNode.liftedTree1$1(Nodes.scala:275)
[error] 	at freechips.rocketchip.diplomacy.MixedNode.x$29$lzycompute(Nodes.scala:260)
[error] 	at freechips.rocketchip.diplomacy.MixedNode.x$29(Nodes.scala:259)
...
```

How can I fix this?

erikdanie, 2019-10-31:
See my branch here where I am attempting to do the same thing. Specifically the diff in  /src/main/scala/everywhere/e300artydevkit/System.scala

https://github.com/sifive/freedom/tree/bump-stuff

yintianyu, 2019-11-01:
@erikdanie I have just switch to your branch besides every submodule. After modifying several import lines(as well as other lines maybe), I could finally generate verilog.

However, when generating mcs, it failed as followed:
```
ERROR: [Synth 8-439] module 'AnalogToUInt' not found [/home/yinty/code/freedom/freedom-bump/builds/vc707-u500devkit/sifive.freedom.unleashed.DevKitU500FPGADesign_WithDevKit50MHz.v:283823]
	Parameter WIDTH bound to: 32'sb00000000000000000000000000000001 
ERROR: [Synth 8-285] failed synthesizing module 'VC707BaseShell' [/home/yinty/code/freedom/freedom-bump/builds/vc707-u500devkit/sifive.freedom.unleashed.DevKitU500FPGADesign_WithDevKit50MHz.v:283675
```

I find `AnalogToUInt` ( along with `UIntToAnalog` ) in `fpga-shells/src/main/scala/shell/Util.scala` where it is a `BlackBox` module.

So where can I find these modules, are they Xilinx IPs? How can I use them when synthesizing?

erikdanie, 2019-11-01:
You need to edit the makefile to have VSRCS variable point to that path

yintianyu, 2019-11-02:
Actually I don't know which path or file to be added to `VSRCS` since I cannot find a verilog module named `AnalogToUInt` in any vsrc folder. I even suspect that it isn't a verilog module.

erikdanie, 2019-11-05:
Ah I see, these were never open-sourced, though they are really simple, and there's no reason they shouldn't be. The reason I hadn't done it yet is that IOBUF is preferred to AnalogToUInt and UIntToAnalog, but I hadnt gotten around to replacing them yet. (https://github.com/sifive/fpga-shells/blob/master/src/main/scala/ip/xilinx/Unisim.scala#L235)

Here is the code if you care to use it though:

module AnalogToUInt (a, b);

   parameter WIDTH = 1;

   inout [WIDTH-1:0]  a;
   output [WIDTH-1:0] b;

   assign b = a;

endmodule

module UIntToAnalog (a, b, b_en);

   parameter WIDTH = 1;

   inout [WIDTH-1:0] a;
   input [WIDTH-1:0] b;
   input             b_en;

   wire [31:0]       z32;
   
   assign z32 = 32'hZZZZZZZZ;

   assign a = b_en ? b : z32[WIDTH-1:0];

endmodule

yintianyu, 2019-11-06:
That works! Thank you very much.

This issue should be closed now.

BTW, can you help me with #144 ?

closed 2019-11-06 by yintianyu