sifive/freedom #147: Using VC709 instead of VC707

abdelrahmanhussein, 2019-11-24:
Hello,

I am totally new to the RISC-V domain. I considering implementing Rocket Chip on FPGA, and I found out that this is doable using SiFive Freedom.

However, it is stated that the supported boards are ARTY and VC707. I am wondering if it is possible to use Xilinx VC709 instead of VC707? If yes, what issues or challenges should I expect out of this switching?

Thanks.

brucehoult, 2019-11-25:
If you have experience with Xilinx FPGAs and Vivado then it should be fairly easy to make it work on any other Xilinx board that has sufficient resources. I believe it's mostly just a matter of remapping the I/O pins from one of the supported boards to whatever the new board has.

I expect we'd be happy to take user contributions of support for other boards, though we wouldn't be in a position to test or support them ourselves.

erikdanie, 2019-11-25:
Yes, @brucehoult is correct, the main difference should be the pin constraints, which are conditionally set in the fpga-shell. If you only want a one-off build, you can just `make verilog` and replace the .xdc with your own, but we will also take a pull request to add support, but with no future support from our side.

abdelrahmanhussein, 2019-11-25:
Thank you @erikdanie  and @brucehoult.

I may have an irrelevant question, that I look really hope you would help me with.

I believe that this [GitHub repo](https://github.com/ucb-bar/fpga-zynq) is going to be useful to me. I know it is no longer maintained, however, I am still wondering if it is efficiently functional with the current release of the [Rocket Chip](https://github.com/chipsalliance/rocket-chip).

Thanks.