sifive/freedom #149: SD Card is showing strange behavior

Iripi97, 2020-02-13:
Hi I am trying to boot Linux with the Freedom Unleashed 500 work flow. I ran into some issues with formatting my SD card. I follow the steps found in the instruction [PDF](https://sifive.cdn.prismic.io/sifive%2Ffa3a584a-a02f-4fda-b758-a2def05f49f9_hifive-unleashed-getting-started-guide-v1p1.pdf) but when I try to write the table to the disk and exit it says:
"Warning! An error was reported when writing the partition table! This error MIGHT be harmless, or the disk might be damaged! Checking it is advisable."
I cannot see it when typing **sudo fdisk -l** either, however I can see in /dev the SD card (it is /dev/sdb). 
How would I check to see if the SD card is damaged, or how can I fix it such that it can still boot Linux?

erikdanie, 2020-02-13:
I am not familiar with fdisk, but you may want to try asking on the [sifive forums](https://forums.sifive.com/) to see if anyone else has more experience or has run into this before

Iripi97, 2020-03-06:
If anyone else has trouble with their SD card, they should try the steps listed here:
https://forums.sifive.com/t/sd-card-is-showing-strange-behavior/3333
On my end it ended up being that the SD card reader was broken, hope this helps!