sifive/freedom #152: QSFP-Ethernet on VCU118

manox, 2020-05-08:
Hello,

I use Freedom unleashed (branch bump-stuff) on a VCU118 with freedom-u-sdk Linux. 
I don't have an FMC Ethernet card, and since there are QSFP overlays for VCU118, I wonder if I can use them for Ethernet under Linux. A first test has shown that at least it does not seem to work out of the box. I would be very happy about any help. Thanks!

erikdanie, 2020-05-11:
Hello,

Unfortunately, the QSFPs cannot be easily be used for ethernet because no open-source controller has been onboarded and connected

manox, 2020-05-19:
Okay understand, thanks!

manox, 2020-07-22:
I looked into the topic again and saw that there is already the Xilinx xxv ethernet MAC IP (for which I also have a license). Is there a way to connect it in unleashed? Thanks again!