sifive/freedom #153: Use Boom core in Freedom

manox, 2020-05-19:
Since the interfaces of Rocket and Boom are compatible, I wonder if it is possible to replace the Rocket with the Boom in Freedom Unleashed.
Let's ask the hypothetical question to the experts, is that possible? Or are there obstacles that make it impossible? Would changes to the Bootrom be necessary?

austinharris, 2020-05-19:
I hacked this together at one point, but it is certainly out of date now:
https://github.com/austinharris/freedom/tree/boom

tmagik, 2020-05-20:
@austinharris that looks very interesting, what might it take to persuade you to update it?

@manox regarding changes to bootrom, I once started on trying to replace the boot rom with U-boot ( https://github.com/tmagik/freedom/tree/bootrom_hacks ) , although in retrospect it might have been better to just port the SiFive FSBL instead.

manox, 2020-06-11:
@austinharris Thanks, this is a good starting point for me!

@tmagik I was wondering if you can start the linux with the boom the same way as with the rocket. it works with the above mentioned repo.