sifive/freedom #167: Where is Class “DevKitU500FPGADesign_WithDevKit100MHz”？

liyewen521, 2020-12-15:
@mcd500 Hello, first of all, thanks for adding support for the no-PCIe version on vc707. I generate Verilog and bitstreams using the following command successfully.
```bash
make -j16 CONFIG=DevKitU500FPGADesign_WithDevKit100MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog
make -j16 CONFIG=DevKitU500FPGADesign_WithDevKit100MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs
```
And now, I want to make some changes to the Chisel code. In rocket-chip, the "CONFIG=xxx" refers to a class in "src/main/scala/system/Configs.scala", so I can change it easily. But in this repository, I can't find a class named "DevKitU500FPGADesign_WithDevKit100MHz".

I'm confused about this, can you give me some advice? Thanks a lot!

mbs0221, 2021-01-27:
Maybe you should have a look at **VC707BaseShell** first in this file ./fpga-shell/src/main/scala/shell/xilinx/VC707NewShell.scala!

liyewen521, 2021-02-16:
Okay, thanks, I found it.