sifive/freedom #171: This project is incompatible with the Arty A7 35T

lindemer, 2021-02-19:
The [Getting Started Guide](https://sifive.cdn.prismic.io/sifive%2Fed96de35-065f-474c-a432-9f6a364af9c8_sifive-e310-arty-gettingstarted-v1.0.6.pdf) does not state that this project won't build for Arty A7 35T Rev. E, which uses a s25fl128sxxxxxx0 memory model. Freedom targets an earlier revision of the same board (Rev. C, I think) which uses the mt25ql128 memory model. It may be helpful to either clarify this or add support for Rev. E. (I know I'm not the first person to have purchased the wrong board.)

AntonHermann, 2021-02-20:
Just came here to open an issue, without knowing what exactly the exact problem was. Are there any workarounds?

AntonHermann, 2021-02-20:
Where do you find the revision? I can't find any markings on the packaging or the board itself. There is a small sticker with "QC.OK", a line and below it the number 2, but I'm not sure what this means and whether it could be related to the board revision.

lindemer, 2021-02-21:
On mine, it's written just below a bar code sticker on the underside. If you're following the guide, you'll know you have the wrong board when you try to flash the .mcs file in Vivado.

erikdanie, 2021-03-01:
You should still be able to use the .bit file, which does not touch the flash. However, the fix would be relatively easy, I imagine just this line would need to change https://github.com/sifive/fpga-shells/blob/14297af2878dc648ffd5751010fa72094ff444b0/xilinx/arty/tcl/board.tcl#L3